Autodesk/synthesis #1: Version controlling unnecessary files

CobaltCake, 2016-08-16:
For example, there are a disappointing amount of `*.dll` files, `obj`, and `bin` directories. These should not be tracked by git, as it's main purpose is *source* version control, not *binary* version control. It's likely that a discussion about `.gitignore` files should occur.

CobaltCake, 2016-08-16:
I've assigned @lessthantrue and @HiceS to this, as things like `git log -- $(find . -iname "*.exe")` reported mostly those two.

philiparola, 2016-08-16:
They do have a .gitignore, but Git will still track files that were created before the .gitignore.
Fixing this is simple.  First commit all changes.  Then run
`git rm -r --cached .`
`git add .`

HiceS, 2016-08-16:
In the engine Directory under Unity 4 & 5 we need to find another way for users to get the compiled .dll for the simulator API and remove then from version tracking. Here is the paths for @tRidha @UmairZ @MackinnonBuck 

This can be displayed as a download on the website if need be but cannot be on the git tracking 

* engine/unity4/Assets/scripts/lib/SimulatorAPI.dll
* engine/unity5/Assets/Scripts/SimulatorAPI.dll

CobaltCake, 2016-08-16:
Thanks, @philiparola, but, apparently a few of those dlls (I think just the ones mentioned by @HiceS) are built internally, the source for which is included in this repository already. However, the build instructions for things requiring those dlls are lacking, so that documentation must be written for a complete fix and for this issue to be closed.

philiparola, 2016-08-16:
I also noticed that my command would be destructive to some necessary files because of some of the wildcard names in the .gitignore (ie: the *build* line will ignore crucial filenames).  I would recommend not using what I've provided unless the .gitignore is more specific.

CobaltCake, 2016-08-16:
Thanks for the heads up concerning your suggestion, but misbehavior like that shouldn't need to be a problem, and as such some time should be spent making our `.gitignore`s less vague. I'll assign that to myself and work on it tomorrow.

HiceS, 2016-08-17:
Removed the Libs folder from the engine research directory and included a path to a zip file on our website if we absolutely need it and for some odd reason cannot build it from OpenTK / might be just lazy. @lessthantrue -> resolved

CobaltCake, 2016-08-19:
Alright, I've de-vague-ified the `.gitignores`. The `emulation/fake_wpilib/gazeboCode` directory has enough problems by itself for a new issue to be made about it.

closed 2017-08-30 by thebenjhoward