Autodesk/synthesis #258: Usage with 2019 build system/GradleRIO

andrewda, 2018-10-10:
Absolute newbie to Synthesis here. Is Synthesis currently setup to work with GradleRIO and the 2019 build system? If so, how would I deploy my code to the Synthesis code emulator?

KiraCorbett, 2018-10-18:
Hi @andrewda! To clarify are you wondering specifically how to deploy code to the Synthesis code emulator? To deploy code, you can build your project in Eclipse as you normally would for your robot. Then in Synthesis under the "Emulation" tab, you will need to select the FRCUserProgram that you just built (but be sure to install the code emulator from our website.) Some of our [emulation tutorials](http://synthesis.autodesk.com/tutorials.html#code-emulation) also guide you through this process.

perdlespedales, 2018-10-18:
Is it necessary to do a Debug build for the code to work in the Emulator, or will the normal Deploy build work as well?

copernicium, 2018-10-18:
A normal build will work fine. In our tutorials, we walk-through building in Eclipse because that used to be FIRST's recommended build environment, but any means of building a functional FRCUserProgram binary will work.

andrewda, 2018-10-18:
Ah okay, I guess the building through Eclipse threw me off. Since GradleRIO/VS Code is the new recommendation from FIRST, it'd probably be good to update the documentation. I can submit a PR once I get it figured out. 

Just to clarify, though: I can build the FRCUserProgram jar and just select that in Sythesis?

copernicium, 2018-10-20:
Yes, that will work.

closed 2018-11-09 by copernicium