Autodesk/synthesis #259: Joystick 4 constantly being pressed down, even when no controller is even connected

yamaan93, 2018-10-13:
When trying to map joysticks and buttons from a controller, "joystick 4" is always registering, and overriding all over button presses, making it impossible to use. I thought this was a problem with the controller, but after restarting the app with dead zones on all the sticks so that no input should be read I found that the application itself thinks that "joystick 4" is always being held down, even with no controller connected. this has rendered the application useless if I can't map controller buttons.

HiceS, 2018-10-13:
Thanks for posting this as an issue, we will have someone look into it. I myself haven't encountered this error but we will look into why it might be happening. 

Can you give us a little information on what kind of machine you are using? And what kind of joystick you last used if any.

HiceS, 2018-10-13:
Okay thank you, that's very interesting. Seems like we may have a bug when dealing with Sony controllers. We do not have any of those controllers for testing under the presumption that they have similar drivers to Logitech. I'll let everyone know and see if we can get a fix out asap. 

Someone else has also emailed us about the exact same issue so I'm almost certain it's the PlayStation controller that we are having an issue with.

HiceS, 2018-10-13:
I'll run to BestBuy soon and see what I can get to test with.

yamaan93, 2018-10-13:
thank you so much,

HiceS, 2018-10-19:
Hey Sorry for getting back to you so late, we have confirmed that it is an issue with the ps4 controller and the library we use to store values into our control scheme. This is a priority issue that will be fixed in the upcoming patch.

yamaan93, 2018-10-19:
That's great, I was worried it was a problem with my controller, good to know

yamaan93, 2019-03-05:
So the bug is still here, after the patch by the way

HiceS, 2019-03-06:
Thank you for letting us know, last patch was mainly made to correct small field issues and drastically improve the code emulation interactions. Unfortunately the engine team didn't include many changes to support fixing the control mappings.

KyroVibe, 2020-06-22:
Remedies have been put in place to address this issue. In the next release this should no longer be an issue. Only took 2 years

closed 2020-06-22 by KyroVibe