Autodesk/synthesis #260: Known Issues

HiceS, 2018-10-19:
Hello everyone, we had a discussion about wether to keep our publicly known issues for the release in a forums post or on the git page. For now the git issue page has won.

- [ ] Still an issue
- [x] Fixed

## Version Number (4.2.0.1): [Release Link](https://github.com/Autodesk/synthesis/releases/tag/4.2.0.1)

- [ ] Controller mappings disappear when changing fields
Fix: Restart program

- [ ] PS4 Controllers are having issues with `Joystick4` [Issue posted](https://github.com/Autodesk/synthesis/issues/259)

- [ ] Fusion 360 exporter buttons wont work 
Fix: right click on palette -> showdevtools (instantly fixes problem) [Issue posted](https://github.com/Autodesk/synthesis/issues/168)

HiceS, 2018-11-06:
Controller mappings disappear fixed as of path 4.2.1.0

HiceS, 2019-06-24:
Fusion 360 Problem seems to have been fixed by the fusion team.

KiraCorbett, 2019-08-20:
Controls were updated and restructured quite a bit. It does not seem that Xbox controllers read as Joystick 1.

closed 2022-07-20 by NickBurrell