Autodesk/synthesis #277: Keybinding conflict between robot control & driver station control

flybotix, 2019-01-02:
To reproduce:
1. Set the keybindings for robot control of Player 1 to a typical gamer-style configuration which uses a keyboard and mouse:
 - W = Forward
 - S = Backward
 - A = Left
 - D = Right
2. Set the camera view to "Driver Station", which is a standard match configuration

Problem: When turning the robot left or right, the camera moves left or right in conjunction with the key pressed.

It would be nice to re-bind the driver's station camera binds to any keys, similar to robot control.  In the very least, the key binding UI should provide a warning that these keys are reserved.

Alternatively, the default binds for camera control behind the glass could be the left/right arrows rather than A/D, as a static option in the keybinding UI.

matthew17754, 2019-01-03:
@flybotix Thanks for pointing that out. That is indeed a valid problem. We will look into adding the driver station camera movement controls to the _Control Panel_ for user configuration.

HiceS, 2019-01-03:
Looks like something that might be good for the next patch.

flybotix, 2019-01-04:
Cool, thanks for the replies guys!  I am going to use this for a game breakdown presentation for pre-kickoff tomorrow; I'll keep notes on anything that would be a 'nice to have' as I setup the fields (2013 & 2017).

KiraCorbett, 2019-04-04:
Closing for now and planning on re-implementing control schemes. Thank you for the feedback.

copernicium, 2019-07-31:
Thank you for reporting this issue. As of #381, you will be able to remap all camera controls except for click and drag to rotate the camera during free roam.

closed 2019-07-31 by copernicium