Autodesk/synthesis #281: Can IDs more than 8 possible?

Sundypha, 2019-02-26:
Currently in the Inventor Exporter we're only able to set maximum 8 CAN IDs while our robot uses 16 motor controllers (Talon SRX). Are we limited to these 8 or are we doing something wrong?

copernicium, 2019-02-28:
Thank you for pointing this out to us. The emulator and engine components of Synthesis currently support up to 64 CAN devices like the RoboRIO does, but you're right, the exporter does not allow you to export your model with IDs that high yet. I'll pass this on to our exporter team, since they know better what fixing this entails. 

In the meantime, for a work around you could temporarily try re-mapping half those CAN devices to PWM channels if they're available. I'm sorry this isn't ideal, but it would at least let you interface with your model in the Synthesis simulator. If your main interest is testing your code's response to inputs, you can still run your code as it is now and watch the RoboRIO inputs and outputs. A user-friendly view of this doesn't exist yet, but you can take a look at our [wiki](https://github.com/Autodesk/synthesis/wiki/HEL-Home#diagnosing-and-debugging-issues) to see how you can accomplish this.

Let me know if you have any questions.

KiraCorbett, 2019-04-04:
Adding item on the public Todo list for the following year of development. Closing for now.
https://github.com/Autodesk/synthesis/issues/260

matthew17754, 2019-06-18:
Fixed in #286

Thank you for reporting this @Sundypha

closed 2019-06-18 by matthew17754