Autodesk/synthesis #282: Emulator tab non-functional

WWRC-FRC, 2019-03-06:
I have just downloaded and installed the latest version of Synthesis (4.2.2.0) and then installed the corresponding emulation package.
When I launch Synthesis I see an emulation tab, but when I click it nothing happens. The icons do not change and I am unable to load code or perform any other emulator specific tasks.

copernicium, 2019-03-06:
Hello, 

This is a known issue, and we are working on fixing the installer. It seems to work for some users, but for others there's this problem exactly like you described. The source of the bug seems to be that QEMU is not installed correctly when the emulator installer runs. If you install it manually, everything should work normally. You can find the QEMU installers [here](https://qemu.weilnetz.de/w64/ "QEMU Binaries for Windows (64 bit)"), and the one we use is [qemu-w64-setup-20181211](https://qemu.weilnetz.de/w64/qemu-w64-setup-20181211.exe "qemu-w64-setup-20181211.exe"). 

Thank you for your patience while we fix this, and please let us know if you have any more issues.

matthew17754, 2019-03-06:
Hi @WWRC-FRC

Thanks for reaching out to us. I have a couple questions regarding your configuration:

- What version of Windows are you running?
- Are you running a 64 bit OS?
- Have you tried reinstalling the emulation package? (QEMU should be installed automatically)

Thank you for your patience while we fix this, and please let us know if you have any more issues.

WWRC-FRC, 2019-03-06:
Thanks for the quick response. Very much appreciated.
Sorry for missing critical information!!
Windows 7 64 bit
I have tried re-installing both Synthesis and the emulator package, yes.
I also tried manually installing QEMU as per copernicium's recommendations and that seems to have fixed the issue :)

Side question... What file should I now load into the emulation? The tutorials are for Eclipse and mention "FRCUserProgram" but also imply it is simply the compiled robot code, but nothing I have tried so far will load. Are VS Code programs supported or do I need to try to move the project to Eclipse to compile it for Synthesis?

Thanks.
BR,
Steve

copernicium, 2019-03-06:
The FRC User Program is what your code compiles down to and what you normally upload to run on the RoboRIO. It doesn't matter how you decide to build it as long as it's with the supported WPILib version (v2019.4.1 as of writing this). For C++ projects, Visual Studio Code creates a file with the name frcUserProgram, which is what you would upload. We currently don't support Java programs because of versioning conflicts with WPILib's changes this build season, but we are working on adding support. Java file names are usually the project name followed by ".jar".

WWRC-FRC, 2019-03-07:
Thanks for the feedback.
I know Autodesk provides Synthesis for free, but it would have been nice to know ahead of time that Java simply doesn't work. I have spent a LOT of time trying to get various simulation platforms up and running, to no avail. I have even tried building C++ examples to try in Synthesis but VS Code won't build them for me at the moment (not your issue, I know, but just another frustration).
It is a shame since I think Synthesis could be REALLY helpful. Unfortunately just too late for my team.

matthew17754, 2019-03-09:
Alright, I've restructured the way QEMU installer execution occurs within the emulation installer and it seems to have resolved the issue. @WWRC-FRC Thanks again for bringing this to our attention. Please let us know if you have any other issues with our product in the future and feel free to re-open this issue if you have any further problems regarding the non-functional emulation tab. We wish you and your team good luck in this years game.

closed 2019-03-09 by matthew17754

WWRC-FRC, 2019-03-12:
Thanks for looking at this. I am sure it will help quite a few teams/members.
BR,
Steve