Autodesk/synthesis #433: Could not find the selected robot!

perdlespedales, 2019-08-13:
In many occasions, when I launch Synthesis (Windows 10 - 64 bits version), the Robot Selection pop-up window opens with the following message: "Could not find the selected robot"
When selecting "Change Robot Directory", I am taken to the correct folder. By selecting the folder again, Synthesis becomes capable of finding the robot.

[image]

KiraCorbett, 2019-08-13:
Thank you for pointing that out. I believe we are loading the incorrect "error screen." It should be possible to still load your robot although our user message is misleading.

perdlespedales, 2019-08-13:
Hello Kira,

You are correct. I typically just click on the already highlighted folder
and Synthesis then proceeds to load the robot correctly.

Cheers,
Ronald

Marcorse, 2019-08-13:
I have the same issue on the opening. I tried what Ronald just says, but it still doesn't work. Is there another way to fix this or to pass through the problem? I'll continue to look for any possibilities. 

Marc

KiraCorbett, 2019-08-14:
Thank you. This issue has been updated and should be available in the next release.

closed 2019-08-14 by KiraCorbett

matthew17754, 2019-08-14:
#438 for reference

yanmannqc, 2020-01-07:
I have the same issue with the latest build. And I can't use the samble or dozer robot at all.