Autodesk/synthesis #471: Error installing on Ubuntu 19.04

cjdenio, 2019-11-01:
**Describe the bug**
While trying to install Synthesis on Ubuntu I got this error: `./SynthesisLinux4.3.1.1.run: 1: eval: ./LinuxInstaller.sh: not found`. It's really not lying (that file doesn't exist), but I think it should be working.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the terminal
2. Run `chmod +x SynthesisLinux4.3.1.1.run`
3. Run `./SynthesisLinux4.3.1.1.run`
4. Observe error

**Expected behavior**
I expected it to install successfully.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 19.04

**Additional context**
¯\_(ツ)_/¯

HiceS, 2019-11-02:
Thanks for letting us know we will spin up a new image and try it out.

I'm really sorry you're having an issue. The Linux support is new for us this year so we are still figuring it out.

cjdenio, 2019-11-02:
That's completely OK! I just moved to Linux, and I was delighted to see that you guys are working on it. It's not a huge priority.

morgajel, 2019-11-14:
Not sure if it's the root cause, but it looks like the LinuxInstaller.sh has DOS linefeeds.

morgajel, 2019-11-14:
```
bash Synth.run --accept --target ./foo/
```

which eventually produces this:
```
++ ./LinuxInstaller.sh
./Synth.run: ./LinuxInstaller.sh: /bin/bash^M: bad interpreter: No such file or directory
+ res=126
```

Which correlates with my previous finding.
I then ran dos2unix on the shell files:
```
$ cd foo/
$ dos2unix LinuxInstaller.sh ProgressBar.sh SynthesisUninstaller.sh 
dos2unix: converting file LinuxInstaller.sh to Unix format ...
dos2unix: converting file ProgressBar.sh to Unix format ...
dos2unix: converting file SynthesisUninstaller.sh to Unix format ...
```
Then ran the installer via sudo:
```
$ sudo ./LinuxInstaller.sh 
Installing Autodesk Synthesis 4.3.1.1 For Linux
This software is licensed under the terms and conditions of the Apache 2.0 License.
Creating Directories...
Initializing Directory Permissions...
Copying Robot Files...
    627,283,392 100%  357.98MB/s    0:00:01 (xfr#151, to-chk=0/165) 
Copying Field Files...
     83,370,419 100%  344.06MB/s    0:00:00 (xfr#39, to-chk=0/53) 
Copying Program Files...
    223,113,171 100%  351.07MB/s    0:00:00 (xfr#207, to-chk=0/225)  
          9,123 100%    0.00kB/s    0:00:00 (xfr#1, to-chk=0/1)
            222 100%    0.00kB/s    0:00:00 (xfr#1, to-chk=0/1)
            552 100%    0.00kB/s    0:00:00 (xfr#1, to-chk=0/1)

Autodesk Synthesis 4.3.1.1 For Linux Installation Complete
```
and then finally ran the binary:
```
/usr/share/Autodesk/Synthesis/Synthesis.x86_64
```

Which produces a fullscreen error message saying the "Robot Instance [is] not valid."

I'm guessing the robot files may have the same DOS line endings problem- I'll check that next.

akrantz01, 2019-11-15:
I'm running into this issue as well on both Ubuntu 18.04 and Manjaro. At the moment, I am getting the error:
```
./SynthesisLinux4.3.1.1.run: ./LinuxInstaller.sh: /bin/bash^M: bad interpreter: No such file or directory
```

HiceS, 2019-11-15:
@morgajel That's awesome and maybe I can shed some light on what I think might be the issue.

So we have to get everything digitally signed to release it in our downloads, since Autodesk does not release almost any other linux software I'm thinking the digital signature software we have might change the line endings that may invalidate the script. It's also possible that since we develop and then publish to git through a git control client it might have asked one of the developers about line endings and they could have accidentally selected the wrong ones.

Matt and I had a conversation about this Tuesday and he's going to look into it this weekend. He thinks it may have something to do with an extra carriage return line that s causing some issues on his end as well. 

If you do find some solutions 100% fork and make a PR we love that stuff <3

@akrantz01 ah dude I'm sorry, I actually don't run Manjaro at the moment but it seems like they might run a custom interpreter or , ~~its possibly they may only use a custom one~~ Seems like bin/bash exists but you may have to use **chsh** to figure out how to use it with that script temporarily. It also seems like there is a special character that is wrong in the install script. I currently run a custom interpreter as well I'm thinking a solution to most of this might be just writing out a file simply explaining what the script does so we can troubleshoot easier that way and we get less hangs from that end and makes it easier for people to submit PRs. Ultimately it doesn't do much other than simplifying the install process slightly.

Using chsh is not a solution for unix based systems to properly install stuff ever. I'm not even certain that is the issue but try this for now and we can add edge cases for popular shells.

` ps -p $$ `

This is on my OSX box but you all seem like you get the idea,
I'm looking for bin/(shell)

[image]

Let me know what that says.

HiceS, 2019-11-15:
@morgajel 

I just saw the additional information you posted about the robot instance not valid error and yeah it's certainly possible the line endings are causing some issues but the more likely scenario is that they are not in the absolute correct file location. We have an awfully strange way of dealing with files that we are currently fixing by streaming them over a webserver, so hopefully that will fix some issues but that's not planned to be integrated until summer so for now try to change line endings and make sure the file exists in the current directories. Also make sure to change line endings of both fields and robots and only the XML / JSON files and ~~not the binaries~~ maybe the binaries too if the carriage return was in-fact an issue.

cjdenio, 2020-01-10:
I've successfully installed Synthesis 4.3.2 on Ubuntu. :tada: It's still buggy, but it works. It would be great if the documentation said where Synthesis is installed, though (I had to look at the installer code). It's `/usr/share/Autodesk/Synthesis` for anyone curious.

closed 2020-02-03 by cjdenio