Autodesk/synthesis #480: No network multiplayer?

cjdenio, 2020-01-10:
**Description**
Has the Network Multiplayer been removed from Synthesis? I can't find it :thinking: 
Thank you!

matthew17754, 2020-01-16:
Hi @cjdenio 

You are correct, the Network Multiplayer feature has been removed, but only temporarily due to some restructuring we have been making to the product. We are planning to re-implement that feature in the upcoming summer, so stay tuned.

If you _really_ want to mess around with Network Multiplayer, you could try downloading [4.2.2](https://github.com/Autodesk/synthesis/releases/tag/v4.2.2.0), as that version still supported Network Multiplayer. Please keep in mind however that this feature was in BETA at that time, but it should still work if both clients download that version.

cjdenio, 2020-01-16:
Thank you, @matthew17754 ! I wasn't sure if I was missing something. :grin: I can't wait to see it again in the future!

closed 2020-01-16 by cjdenio