Autodesk/synthesis #482: Qemu not found on Arch

macmv, 2020-01-16:
**Describe the bug**
On Arch, I have installed the 'qemu', 'qemu-arch-extra', and 'samba' packages, but Sythesis cannot find qemu. I don't know if there are more packages I need to install, but no combination of these will work.

**To Reproduce**
Steps to reproduce the behavior:
1. Install 4.3.2 through the latest github release for linux.
2. Install 'qemu', 'qemu-arch-extra', and 'samba' through pacman.
4. See error

**Expected behavior**
The emulator tab should show a Start button, not a Not Installed button.

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Version 4.3.2

matthew17754, 2020-01-16:
Hi @macmv 

Unfortunately there has been an update to the WPI Library API, and as a result, our emulation system is down at the moment. I do want to note that currently the only officially supported Linux distribution is Ubuntu.

I would recommend running emulation using the previous version of the WPI Library on a Windows machine. We apologize for the inconvenience.

closed 2020-01-16 by matthew17754

macmv, 2020-01-16:
Thank you for the quick response. Do you have any estimate of when the emulator will work again?

matthew17754, 2020-01-16:
We are planning to do some restructuring to our current emulation system, due to the maintenance overhead of WPI constantly restructuring their API. I would expect to have the new system implemented by the end of this summer. Again, I'm really sorry about this.

macmv, 2020-01-16:
Thanks for the information. I'm glad we could get this cleared up.