Autodesk/synthesis #483: Emulator support for macOS

wisaac407, 2020-01-26:
**Is your feature request related to a problem? Please describe.**
Can't run the emulator on macOS.

**Describe the solution you'd like**
To be able to run the emulator on macOS. It looks like this has come up before (#477), so I'm wondering if there's a timeline for when this is expected to be implemented.

HiceS, 2020-01-26:
Hey @wisaac407 ,

Sorry I just woke up, we are actually planning some neat changes to the simulator you may be interested in. We haven't finalized the changes that we will be making but the expected timeline is that we release an early edition in the first half of the summer.

We are hard at working making prototypes for how emulation can help people in the future and currently we are defining an event system in a new engine that will be able to handle a generic set of api calls that anyone can write regardless of FRC library. The ultimate goal is to allow an extension of this api so that frc wpilib code can be translated to it easily. I was actually working on this pretty late last night and it's looking good. Currently everything we are prototyping is closed source because of some rules on how I am allowed to interact with synthesis. 

Hopefully soon we will have a roadmap for everyone and I will try to post it on the github and mention anyone that has suggested improvements so they can weigh in on the proposed changes.

HiceS, 2020-01-26:
I'm going to leave this open until I am able to post a roadmap and get some feedback to make sure that it's the same improvement you are talking about.

The new version has a pretty large emphasis on cross platform interaction as I actually work mostly on my mac.

airsquared, 2020-09-23:
@HiceS any updates on this so far?

HiceS, 2020-09-23:
@matthew17754 @NickBurrell @copernicium

I think you all might be able to explain your changes and share when the beta test for the new emulator version is happening.

copernicium, 2020-10-15:
Synthesis is undergoing a modernization which has removed the emulator present until v4.3.3. There is now a control API for the Synthesis engine, which is how robot code will be able to interact with robot models. This is the Controller module. 

To use the Controller to control a robot, for now, you will need to write a script in Python or C++ that includes the controller library. Then, you can use the API to set motor voltage percents for robot models.

closed 2022-07-20 by NickBurrell