Autodesk/synthesis #484: Error when loading robot code from the 2020 season.

ophbalance, 2020-01-26:
**Describe the bug**
The Robot Code Connected never completes

**To Reproduce**
Build the project from https://github.com/djbrook4/6004-2020-V2 and load the compiled JAR file

**Expected behavior**
simulation will run

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1903

HiceS, 2020-01-27:
This is awesome thank you for the information. We have a pretty large amount of users experiencing this problem and we have found that it is a result of WPI updating the hardware layout for the roborio which means our mappings are invalid. 

We are working on a new version of the Emulator that is expected to release in the first 5 weeks of summer as a standalone API for testing and machine learning. 

Again thank you for the write up and hopefully this can help some people that are also experiencing this issue. You may be able to compile and run the code by installing last year's wpi dependency if you really want to play with the Emulator.

closed 2021-03-31 by matthew17754

matthew17754, 2021-04-01:
Latest emulation updates for WPILib2021.2.2 support have been officially [released](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3.2) in Synthesis4.3.3.2. Thanks for your inquiry.