Autodesk/synthesis #485: MD5 checksum error

thekadenh, 2020-02-08:
I've tried downloading the .run file from your website multiple times on different networks and each gives the following error: 

./SynthesisLinux4.3.3.run 

Verifying archive integrity...  100%  Error in MD5 checksums: 9ad4b1713f8037af0e19d82daac9301d is different from 6ebc204e0021d72674c6bfb35e7451a6

matthew17754, 2020-02-09:
Hi @thekadenh 

Sorry you're having difficulties with the download. 
Try downloading it from [here](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3) and let me know if you experience the same issue.

thekadenh, 2020-02-10:
Hello,

It still gives me the same MD5 checksum error when I download it from there. Do you have any other solutions? Thank you!

matthew17754, 2020-02-14:
Sorry for the delay in response. This is the first time I've encountered this particular issue regarding the checksum. I'll need to do further investigating on this for the time being. My apologies for the inconvenience. I'll let you know if a fix is derived.

motivator, 2020-03-13:
FYI. the problem here is the gzip data in the installer is corrupt, I didn't try creating it myself but that's definitely the issue. might be as simple as trying a rebuild of the installer.

**v4.3.2**
```
$ bash ./SynthesisLinux4.3.2.run --check
Verifying archive integrity... ./SynthesisLinux4.3.2.run does not contain an embedded SHA256 checksum.
  100%   MD5 checksums are OK. All good.

$ tail -n +588 ./SynthesisLinux4.3.2.run | gunzip -c -t -v
 OK
```

**v4.3.3**
```
$ bash ./SynthesisLinux4.3.3.run --check
Verifying archive integrity... ./SynthesisLinux4.3.3.run does not contain an embedded SHA256 checksum.
  100%  Error in MD5 checksums: b4c253dcf200a440c0cba67fa5e82e34 is different from 6ebc204e0021d72674c6bfb35e7451a6

$ tail -n +588 ./SynthesisLinux4.3.3.run | gunzip -c -t -v

gzip: stdin: invalid compressed data--format violated
```

jakbyte, 2020-03-27:
I'm having the same issue on Arch Linux.

jakbyte, 2020-03-27:
I tried the installer in the releases page, and it worked!
However, the installation failed because I didn't have rsync installed.

jakbyte, 2020-03-27:
Where is it installed, exactly?

matthew17754, 2020-03-28:
@motivator Thanks for the input. I've updated the latest [release](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3) to resolve the checksum error.

@jakbyte Interesting note about rsync. Glad you got that to work on Arch. As of right now we've currently only tested Ubuntu. I'd be interested to hear about your success with Arch.

@thekadenh Let me know if this resolves your issue. Sorry for the delay.

closed 2020-04-05 by matthew17754