Autodesk/synthesis #486: The entire screen shakes

TeamPiSoftware, 2020-02-09:
**Describe the bug**
The first time when i start the program after downloading it. It workes "fine". Except the robot flies everywhere. Something with the physics. (when the robot has less parts it works better).
Every time when I start the program after that, the entire screen UI shakes. And I can't click on anything in th UI.

**To Reproduce**
Download it.
open the programm.
close the programm.
and re open it.

**Expected behavior**
good working physics and an UI which stays on it's place

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
-Windows 10
-synthesis 4.3.1.1
-NVIDIA GeForce GTC 1070 max-Q
-i7 processor

**Additional context**
Add any other context about the problem here.

HiceS, 2020-02-09:
That's pretty strange but weirdly enough we have had one other user report the same thing 3 years ago. Before we changed physics engines or any of that.  Is it possible you may have a version of mono that's meant for mobile device development or maybe have some kind of phone emulator turned on?

Sorry the issue form is so long we just thought it might help us get some information on what's happening. 

We had thought that we released a fix for it a long time ago but this night be an issue with unity itself.

My only other question would be what size monitor you have. Is it a specific resolution and dpi that is more unique.

NickBurrell, 2022-07-20:
Closing for relevance.

closed 2022-07-20 by NickBurrell