Autodesk/synthesis #489: 2020 Summer Intern Applications OPEN NOW!

HiceS, 2020-03-04:
# 2020 Synthesis Internship Post

Hey all we are opening our internship application for the 2020 summer internship that extends from:

**__June 22nd 2020 - August 28th 2020__**

(Link to our internship application)[https://synthesis.autodesk.com/internship.html]

Below you can find some information, FAQ, What the devs are looking at, and our new modernization information:

## FAQ

1. What is the skill level required of an intern?
We interview and look for interns in all skill ranges from beginners to super users, my recommendation is if you get excited about robotics simulation we would love for you to apply. Honestly a lot of what makes developers crazy good is their passion to improve a project.

2. What age do I need to be?
You need to be 16 by June 22nd Technically. For any more information or questions regarding this please email frc@autodesk.com.

3. Where are the position descriptions?
Currently we haven't posted the hiring position descriptions and rundowns (by accident) but when they are available which will be soon they will be located on our internship page found on our website in the link above.

4. Where is the internship located?
The internship is located in Portland, Oregon. You need to present during the internship in the Portland office which can be difficult for students that need to find housing so we try to make decisions early in the year if possible.

(Any other questions we get from emails or forums or in person should be added here to be tracked if viable for general public.)

## Development Ideals

We are actively trying to find people that fit into both specific and general development roles. Specific meaning that some positions require advanced knowledge of a topic in technology we are trying to take advantage of to modernize our software.

Some of the things we are currently investigating for modernization:
 - Compute Shaders and Graphical Parallelism
 - GRPC
 - Protobuf
 - Physx in Unity
 - Customer Deliverable

Many of these we need to refine, we have built some prototypes for the new engine and exporter as well as the basis for what we are calling the "Controller". Which is a replacement and enhancement of the emulator. (more in the modernization section)

## Modernization

After looking through our JIRA and debilitation on improving the current state of the engine and physics simulation we have made some difficult decisions surrounding the continuing development of the physics simulator with Unity.

#### Unity
Unity is a fantastic platform and we will continue to use it to develop the simulator after prototyping some modern fixes for older issues. We will be using the most up to date unity edition to redesign basic functionality with better rendering pipelines to improve machine dependent performance issues as well as enabling the integrated PhsyX platform in Unity to do our physics solving. Through the last two years they have made so much progress and enhancements to their built in physics layer that it was an easy decision to abandon the bullet wrapper we were using previously. This will also come with a host of improvements to the code base and completely axe off any asset store dependencies.

#### Fusion 360
From our analytics and communicating with teams it seems that Fusion 360 is increasingly popular with our user base. However this is also the most problematic and hard to maintain feature of our codebase which is why we have switched to developing in C++ to developing in Python in order to reduce complexity and training overhead for new developers. It also supports our goal of minimizing the exporting steps and maximizing customization in Unity for reduced number of "re-exports" to fix the user story of re-exporting the robot to re-configure the CAN values or Gearing values. We also entirely depend on Protobuf to define how we export geometry and join values from the fusion 360 model which reduces save time, memory consumption, and hard drive space for files. Overall we were also driven to make these changes to support the OSX users that have been giving us feedback about the usability of Synthesis on other machines.

#### Inventor
We are still deciding on Inventor changes and support may be delayed until finalizing changes for Fusion 360 and supported exporting formats to minimize old file error issues between the two.

#### Controller
The most exciting new change for us is the Controller product. The Controller will feature a set of communication protocols in order to enable outside developers to use a common communication library to control the robot and world simulation. One of the development tasks is also to create a template repo that contains the basics of how to develop a new library in any language you choose to communicate with Synthesis over GRPC. We will be developing a sample library to use for those users who want to develop code for robots and practice with a variety of different libraries in a simple way. This is going to be a main focus for us in creating new experiences for our users in ways the hopefully encourage learning and possibly creating workflows we previously hadn't thought of. We are very excited to see what we can come up with in this area, we are also looking for developers in this area with experience in using libraries in other languages and understand async code as well as GRPC.

## Recommendations:
 - Put your github on your resume (please)

 - We are really interested in hearing about your projects and the cool things you do for fun. I find a lot of people will forget to include fun projects in resumes or applications but at Autodesk we love to see awesome people making cool things so don't feel afraid to include too much information.

 - We won't care how long your resume is as long as you're using the space constructively and it can describe what you like and what you have done, so please don't delete anything for the sake of saving a page. Especially if it's technical.

- We don't want super short ones, many people miss out on opportunities because they must either think we won't look at the resume or we wont want to look at a long one but I promise you we all look through all of them and most of us are developers so please include some technical stuff. (Personally I care a lot less about where you go to school and far far more about when you used opencv to do some cool things with graphics or when you wrote a world of warcraft script etc. so please don't only put your school give us something to get excited about for meeting you)

HiceS, 2020-03-04:
all teh labels 👍

matthew17754, 2020-05-20:
Applications are now closed.

closed 2020-05-20 by matthew17754