Autodesk/synthesis #491: Can't run Synthesis on Arch Linux

jakbyte, 2020-03-27:
**Description**

After installing Synthesis on my machine and looking through the installer script for the installation location, I tried to run it.

```
~ ➜ /usr/share/Autodesk/Synthesis/Synthesis.x86_64
fish: The file “/usr/share/Autodesk/Synthesis/Synthesis.x86_64” is not executable by this user
```

So I tried running it again with `sudo`...

```
~ ➜ sudo /usr/share/Autodesk/Synthesis/Synthesis.x86_64
Found path: /usr/share/Autodesk/Synthesis/Synthesis.x86_64
```

A blank fullscreen window appears, and...

```
fish: “sudo /usr/share/Autodesk/Synthe…” terminated by signal SIGSEGV (Address boundary error)
```

I didn't get any more output, so I have no idea what the issue could be.

**System Information**

* Operating System - Arch Linux
* Desktop Environment - GNOME 3.36.0
* Window Manager - Wayland 1.18

matthew17754, 2020-03-28:
This is caused by the same issue in #485 **MD5 checksum error**

I've updated the latest [release](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3) to resolve the checksum error which should fix this.

I will note that the only currently supported Linux distribution is **Ubuntu 64 bit**, but you're welcome to try other distributions.

closed 2020-03-28 by matthew17754

jakbyte, 2020-03-28:
It installed properly this time, but it still only has a black window that immediately closes.

matthew17754, 2020-03-29:
@jakbyte Is the black window generated by Unity or the shell installation?

In other words, does this occur when launching the Synthesis application or as part of the installer?