Autodesk/synthesis #494: Synthesis installer skipping directories when installing

FlynnDuniho, 2020-04-19:
**Describe the bug**
When I install Synthesis, using the 64-bit installer provided on the Autodesk website, with all boxes checked (including the box to install the Fusion 360 addin), the Synthesis default robots and the Fusion 360 exporter addin both are not installed. 
C:\Users\<USER>\AppData\Roaming\Autodesk\Autodesk Fusion 360\API\AddIns
path is empty, I also had other addins there, but they were deleted in the process of uninstalling Synthesis, I think. I also don't have a Synthesis folder anywhere in my AppData folder. I have started the program multiple times, but never got past the starting screen where I select my robot, because I don't have any robots, default or otherwise. I am using an installer downloaded three days ago.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the 64-bit version of the Synthesis installer
2. Run the installer

**Expected behavior**
I expect the installer to install the sample robots and Fusion 360 addin

**Screenshots**
C:\Program Files\Autodesk\Synthesis\Exporter
[image]

This is where I found what looks like to be an Inventor addin, maybe? These are all of the files in that directory.

**Desktop (please complete the following information):**
 - OS: Windows 10
Version 1909

matthew17754, 2020-04-19:
@FlynnDuniho Can I get a screenshot of your `%AppData%\Autodesk` folder? Make sure the file path at the top is also visible in the image.

FlynnDuniho, 2020-04-19:
Sure thing: This is the path: C:\Users\Flynn\AppData\Roaming\Autodesk
[image]

FlynnDuniho, 2020-04-19:
The ApplicationPlugins actually has my other Fusion 360 addins, just not the Synthesis one

matthew17754, 2020-04-20:
Very unusual indeed. I have a couple more suggestions:

1. Try downloading the installer from [here](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3).

2. Try running the installer as Administrator.

Let me know if these yield any luck.

FlynnDuniho, 2020-05-03:
Okay. Sorry this took so long. Logged in as admin, it correctly generated the plugin files in the Admin AppData, and it generated the default robots in the admin AppData. I then copied those files into my own user's AppDatadirectory, and the Fusion 360 exporter plugin appears in the toolbar. I have yet to test actually exporting a robot, but it looks like the functionality is there. I also opened Synthesis, and successfully loaded a default robot, after changing the robot path from the C:\program Files\Synthesis to the Synthesis AppData robots folder. Everything now appears to work properly. Thank you for your help!

matthew17754, 2020-05-03:
Glad I could be of service.

closed 2020-05-03 by matthew17754