Autodesk/synthesis #495: post-install.sh on MacOS shouldn't rely on homebrew

Lugal-PCZ, 2020-05-12:
**Is your feature request related to a problem? Please describe.**
Most users don't have homebrew installed, so post-install.sh shouldn't rely on it.

**Describe the solution you'd like**
Use the tools that are part of unmodified MacOS installs.

**Describe alternatives you've considered**
1. Remove the "Sanity check for brew" block.
2. Replace `wget https://github.com/Autodesk/synthesis/releases/download/v4.3.3/SynthesisSampleAssets.zip -q --show-progress` with `curl -LO https://github.com/Autodesk/synthesis/releases/download/v4.3.3/SynthesisSampleAssets.zip`

matthew17754, 2020-05-12:
@Lugal-PCZ Thanks for the advice. If you have time, a pull-request would be much appreciated. =)

HiceS, 2020-05-13:
@Lugal-PCZ As part of our future development and restructuring we will remove the need for the post-install script entirely. I apologize for the brew reliance at the time we discussed it as a pre-condition for the emulator to work in order to install some of the emulation dependencies without having to rebuild them. But for sure we will remove that.

HiceS, 2020-05-13:
On that note we are trying to make an addition to the engine rewrite that would allow LUA scripts to execute as part of the main event loop and on start which would ideally replace any fetch functionality of prior scripts if you have any suggestions im always excited to hear them.

closed 2022-07-20 by NickBurrell