Autodesk/synthesis #674: Would it be possible to install the RobotExporter Addon For Fusion360 manually.

R0NAM1, 2020-10-26:
**Description**
_What's going on?_

On My Mac (OS 10.15.6) I am installing the latest Synthesis version (4.3.3), and when I load up Fusion 360 (2.0.9142, we use specific versions on the admin installer for project compatibility reasons), I never am able to install the FusionRobotExporter, it never shows up in the add-ons menu.

Is it possible to install the add-on separately from synthesis, or is this planned for the future? 

**Screenshots**
If applicable drag them in here

Thank you for submitting this, it will automatically assign one of our developers to help you out.

R0NAM1, 2020-10-29:
After some looking around I found out in the extensions directory there exists the glTF exporter, written in Python. After installing it, it is unable to find/download the dependencies it requires:
`    pipDeps = ["pygltflib", "numpy", "protobuf", "pyquaternion"]
`
It seems to stem from the fact that it requests the downloader to use pip, not pip3, which means the addon uses python 2.7.
`        subprocess.run(f"\"{pythonFolder / 'python'}\" -m pip install {depName}", shell=True)
`
It seems to fail at this step, I will debug more.

HiceS, 2020-10-29:
@R0NAM1 I actually use that script as well for part of the internal Autodesk addins we make, can you tell me more about the specific versions of Fusion 360 you use? Fusion installs it's own version of Python which that subprocess call is utilizing in order to fetch pip and install the dependencies. I verified it works well on my mac but I suppose if you are forcing fusion to maintain an older version then it's possible the version of python isn't updated. However I don't know how you would force fusion to maintain an old version. If the addin doesn't show up at all in the addins window then there is a bigger problem then being able to load as fusion doesn't recognize it as an addin. Maybe you can clarify a bit?

R0NAM1, 2020-10-29:
Due to project file incompatibilities, we use a non-auto updating version of Fusion 360 (Currently 2.0.9142), I assumed the add-on would be in the add-ons folder.

R0NAM1, 2020-10-29:
It also seems after it runs 
`subprocess.run(f"curl https://bootstrap.pypa.io/get-pip.py -o \"{pythonFolder / 'get-pip.py'}\"", shell=True)`
The file does not appear in 'pythonFolder' which is 'PosixPath('/Applications/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/bin')'

R0NAM1, 2020-10-29:
After manually copying get-pip.py to 'pythonFolder' (/Applications/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/bin) it correctly finds the file and executes the commands to install the deps. Issue is likely access related since I needed to use sudo to copy the file into the Fusion app.

R0NAM1, 2020-10-29:
Just realized the glTF exporter is not the Robot Exporter, as far as I can tell, don't know why I closed it.

HiceS, 2020-10-29:
Looks like the current version I have pulled down on the release streamer of Fusion is (2.0.9305) so your version should be fairly updated.

However it's worth noting the version you supplied has some bugs that were hotfixed the same day it came out that have to do with the Python api specifically.

https://forums.autodesk.com/t5/fusion-360-api-and-scripts/all-my-plugins-now-don-t-work-with-newest-update-october-2020/td-p/9812006

I've tested this on a few different osx machines and they all seem to work fine and we cannot reproduce the bug. I actually work on the Fusion API team and I have never heard of an addin running while not in the addins menu and that sounds like a very dangerous game.

I would suggest that the problem is external to this addin and is more likely based on whatever distribution that Fusion 360 came to you with. The Autodesk app package and distributions very specifically configure many things during install and update. 

Until someone else sees this problem I'm not super sure what any of us can do to help with this problem, Autodesk is pretty clear that turning off auto updating in Fusion is against the expected user workflow which makes it difficult to accommodate this kind of workaround.

#### GLTF

This version of the exporter has to do with the Synthesis team upgrading the way the engine and exporter works to suite the new technology and to improve the performance of the simulation. I think for this version the expected timeline to get it to users is the end of December so that is when I would expect a large overhaul with significantly better exporting and engine side mechanics.

The other version is on effectively LTS until one of the members decides it's not from what I understand.

closed 2022-07-20 by NickBurrell