Autodesk/synthesis #675: My emulator won't show 'Robot Code Connected'

TrevCan, 2020-11-02:
**Describe the bug**
After installing the latest version of Autodesk Inventor Professional 2021 and Synthesis, my code will not run. The status of each process in the driver station eventually turns green except for **Robot Code Connected**. I believe this is because the emulator is running an outdated version of the WPI libraries, I have the 2020.1.2 version. The driver station log file says that the emulator has the 2019.4.1 version. I have already tried changing up the code and just setting a certain power to motors of the code but none will work. I am using java (JDK 11).

**To Reproduce**
Steps to reproduce the behavior:

0.1. Download [SimpleChassisProject.zip](https://github.com/Autodesk/synthesis/files/5472431/SimpleChassisProject.zip), extract it and build it using gradle (e.g. `gradle build`).
1. Open Synthesis (4.3.3.0)
2. Select a robot
3. Go to Emulation
4. Select "Start Emulator" Wait for the 'Emulator Connected' green status light located on the Driver Station.
5. Select "Select Code" and select the .jar file (located at `SimpleChassis/build/libs/SimpleChassis.jar`)Wait for the 'Robot Code Uploaded", "Robot Code Running", and "Robot Code Connected" green status lights located on the Driver Station.
**Step 5 is where the bug happens, "Robot Code Connected" never changes to a green status light.**
6. Click on "Use emulation".

**Expected behavior**
it should show all 4 green status lights on the driver station.

**Screenshots**

- Driver Station log
[image]
- Synthesis screenshot

**Desktop (please complete the following information):**
 - OS: Windows 10 Home
 - Version 1909
- Synthesis version 4.3.3.0

**Resources**
- JAR build (I know it ends in .zip but otherwise I wouldn't be able to upload this to a github issue):
[SimpleChassis.zip](https://github.com/Autodesk/synthesis/files/5472406/SimpleChassis.zip)
- Source code (gradle java project):
[SimpleChassisProject.zip](https://github.com/Autodesk/synthesis/files/5472431/SimpleChassisProject.zip)

KyroVibe, 2020-11-02:
Hi @TrevCan . The emulator is still only compatible with 2019 WPILib. For 2020, we decided to not update this and take a different approach still in development. If you want, you can role back your SimpleChassisProject to 2019 WPILib and test it that way. If you want any assistance in that, I could help you modify the current project you're working with.

TrevCan, 2020-11-02:
Thanks for the reply, I would very much appreciate assistance in converting the project back to the 2019 version. I have tried modifying the version in the `build.gradle` file but does not seem to do the trick.

TrevCan, 2020-11-02:
I would recommend stating this in the current documentation / guides on the synthesis website (and also in the YT channel).

KyroVibe, 2020-11-02:
I'd suggest rolling back the WPILib vscode extension to before 2020 (such as 2019.4.2), create a new project with the new version, then move your code to the new project and update it to work with 2019 WPILib.

TrevCan, 2020-11-03:
Hey, I just did that and the code sort of seems to run, since I do get my printlns from `robotPeriodic`, but the "Robot Code Connected" never turns green and I cannot run teleop / autonomous or use the driver station at all.
I created the template from vs code with the 2019.4.2 release:
This is the last code I was testing:
[Sample2019_project.zip](https://github.com/Autodesk/synthesis/files/5478775/Sample2019_project.zip)

KyroVibe, 2020-11-16:
@TrevCan sorry for taking so long on this issue. It seems that 4.3.3 has an odd bug that causes emualtion to break. By using [4.3.2](https://github.com/Autodesk/synthesis/releases/tag/v4.3.2), you can avoid the bug. I've tested this with C++ and Java and have been able to enable, control motors, and print messages to the console.

TrevCan, 2020-11-17:
Thanks @KyroVibe . It worked. I was also wondering, is there any documentation regarding the setup of the motor ports (I understand that the motors are set up in Autodesk Inventor, but wasn't able to find any configs on the ports)? If so, where can I find it? Finally, how can I set up CAN motors?

KiraCorbett, 2020-12-19:
@TrevCan You can check our Synthesis 4.2 tutorials on our [YouTube channel here](https://www.youtube.com/watch?v=Jrfy-aggwVU&list=PLuTlzvthXjDKHh5LUufyyZn3OBuNXFkQ7) or our [Google Codelabs tutorial for setup here](https://synthesis.autodesk.com/tutorials/inventor-tutorial.html#3). For Fusion, please see [these tutorials here](https://synthesis.autodesk.com/tutorials/fusion-tutorial.html#0).

closed 2020-12-19 by KyroVibe

copernicium, 2020-12-25:
@NickBurrell discovered that, in Synthesis [v4.3.3](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3), the file `grpc_csharp_ext.dll` is missing from the installer. As a result, the Synthesis project couldn't find the networking library needed to connect to the emulator. When this file was manually added by copying it over from [v4.3.2](https://github.com/Autodesk/synthesis/releases/tag/v4.3.2), that resolved the connectivity problems for C++ code.

Jesmuc, 2020-12-27:
@copernicium that also works for java code (generated with 2019 WPILib), thanks!