Autodesk/synthesis #678: Linux installer reports mismatched MD5 checksum

briceonk, 2020-11-05:
**Description**
The latest Linux Synthesis [installer](https://synthesis.autodesk.com/download.html) seems to be reporting mismatched MD5 checksums. Not sure if this is an issue on my end or not, but I downloaded the archive 4 or 5 times and got the same data each time.

```bash
$ chmod +x SynthesisLinux4.3.3.run 
$ ./SynthesisLinux4.3.3.run 
Verifying archive integrity...  100%  Error in MD5 checksums: 9ad4b1713f8037af0e19d82daac9301d is different from 6ebc204e0021d72674c6bfb35e7451a6
$ sh SynthesisLinux4.3.3.run
Verifying archive integrity...  100%  Error in MD5 checksums: 9ad4b1713f8037af0e19d82daac9301d is different from 6ebc204e0021d72674c6bfb35e7451a6
```

Configuration: Ubuntu 20.04.1 LTS, x86_64 architecture

matthew17754, 2020-11-05:
Hi @briceonk. Thanks for notifying me of this. Try downloading the executable from the GitHub [release](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3) page and let me know if the issue persists. I'll investigate the issue with the checksum mismatch on the download site. Hope all is well with the team.

briceonk, 2020-11-07:
Thanks, @matthew17754 ! The script from the GitHub release page worked.

closed 2020-11-07 by briceonk