Autodesk/synthesis #703: Bug in the 2020 version of the emulator

Rorybabory, 2021-01-05:
I got a bug on the 2020 version where when trying to run emulator code it gets stuck at robot code running, and never says robot code connected. I swapped to the 2019 version and it worked, however I am wondering what the solution is for 2020, as our team wants to use the newest API.

KyroVibe, 2021-01-06:
Hi @Rorybabory , we are planning to release an update for kickoff which will include support for 2020 emulation. Currently as it stands it will only support c++ but we are hoping to add java support as well. Once java support is done we'll update to 2021 wpilib

closed 2021-03-25 by matthew17754

matthew17754, 2021-03-25:
C++ 2020 support has been implemented and C++ and JAVA support for WPILib 2021 should be released in another week or so.

matthew17754, 2021-04-01:
@Rorybabory Latest emulation updates have been officially released in [Synthesis4.3.3.2](https://github.com/Autodesk/synthesis/releases/tag/v4.3.3.2). Thanks for your inquiry.