Autodesk/synthesis #705: Unable to properly load exported field

GarinHamburg, 2021-03-15:
**Description:**
After Exporting a field design using the Synthesis Exporter, the Field fails to load providing the error <br>"Failed to load field metadata"

**Steps To Reproduce:**
1. Create a field design in Inventor Pro 2020
2. Export it using the Field Exporter
3. Attempt to launch field in Synthesis

**Expected behavior:**
The field was expected to load properly

**Screenshots:**
[image]

**Desktop:**
 - OS: Windows 10
 - Version: 20H2
 - Build: 19042.867
 - Synthesis Version: 4.3.3.1

**Requesting:**
- Any information on additional setup that may be required
- How to Export field metadata properly

KyroVibe, 2021-03-15:
Hi @GarinHamburg , could you please send a copy of your exported field?

GarinHamburg, 2021-03-15:
> Hi @GarinHamburg , could you please send a copy of your exported field?
Here you go
[FieldFinal.zip](https://github.com/Autodesk/synthesis/files/6143980/LiquidLifeMapFinal.zip)

KyroVibe, 2021-03-15:
@GarinHamburg thanks for your report. It would appear the field_meta.xml file was never generated. I don't have access to inventor atm so I can only see what the definiton file tells me, but it looks like the field is empty. Is this intentional or another issue?

KyroVibe, 2021-03-28:
@GarinHamburg have you managed to resolve this issue?

NickBurrell, 2022-07-20:
Closing due to no longer supporting legacy Inventor Exporter

closed 2022-07-20 by NickBurrell