Autodesk/synthesis #827: Be able to use assemblies from other files and not have it break

Spaceman113138, 2023-06-01:
**Is your feature request related to a problem? Please describe.**
If you use assemblies that are not part of that fusion file when you open the game, it goes crazy and all the parts lie in ground and shake.
Are you experiencing an issue? Do you have a good way it can be fixed, please tell us.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for requesting a new feature, we take these very seriously and they are all added into our end of year planning for the following year's development cycle.

KyroVibe, 2023-06-01:
### Update
Problem lies with Fusions entityTokens and them being reused across design documents. As a result all instanced bodies of similar root parts are added under one root part. This bug is planned for Sprint 1

KyroVibe, 2023-06-17:
Bug was addressed #831 , will be seen in next release.

closed 2023-06-17 by KyroVibe