Autodesk/synthesis #948: Add-in not working

kappa95, 2023-10-26:
**Describe the bug**
Installed Synthesis V6 from website but the Add-in does not appear in Inventor Professional 2024 and in Fusion 360 is not opening at all.

from Synthesis log file:

```
2023-10-26 11:16:46,369 - synthesis.import_manager - ERROR - Failed
Traceback (most recent call last):
  File "C:\Users/inse9/AppData/Roaming/Autodesk/Autodesk Fusion 360/API/AddIns/Synthesis\src\general_imports.py", line 36, in <module>
    from proto import deps
  File "C:\Users\inse9\AppData\Roaming\Autodesk\Autodesk Fusion 360\API\AddIns\Synthesis\proto\deps.py", line 3, in <module>
    from os import __file__, path
ImportError: cannot import name '__file__' from 'os' (unknown location)
```

KyroVibe, 2023-10-26:
Hello, quick note, we no longer have an Inventor Exporter, however the issue with the Fusion 360 addin is currently being looked into right now.

KyroVibe, 2023-10-26:
@kappa95 Could you tell me what version of Fusion you are using?

kappa95, 2023-10-26:
The version of Fusion is: Fusion 360 2.0.17457 x86_64. On OS: Windows 11 Home 22H2 (22621.2428)

KyroVibe, 2023-10-26:
@kappa95 could you try starting the addin in debug mode. For some reason that seems to fix the issue

kappa95, 2023-10-26:
It opens VSCode and if I press Run i got the following error:
```shell
Exception has occurred: ImportError
attempted relative import with no known parent package
  File "C:\Users\inse9\AppData\Roaming\Autodesk\Autodesk Fusion 360\API\AddIns\Synthesis\Synthesis.py", line 1, in <module>
    from .src.general_imports import root_logger, gm, INTERNAL_ID, APP_NAME, DESCRIPTION
ImportError: attempted relative import with no known parent package
```

kappa95, 2023-10-26:
I have stopped from Addin. And restarted normally, and now the icon in utilities appears. Probably It was missing a local version of Python (for Autodesk I mean) and then It was needed the requirements i suppose?

kappa95, 2023-10-26:
Trying to export now I get the following error: [image]

KyroVibe, 2023-10-26:
@kappa95 Okidoki, definite progress. From the error messages it looks like one of your joints is corrupted. This can happen depending on a lot of things (most of which are issues with Fusion 360 itself). Considering it looks like you only have 3 joints, I would try to rejoint them and see if that solves it. In the meantime, could you send me a sharelink to your design so I can take a look for myself?

kappa95, 2023-10-26:
Yes sure! Here it is: https://a360.co/473xMKE

kappa95, 2023-10-26:
I have tried to save another file and remove old joints and reputting them, but the error is the same.

KyroVibe, 2023-10-26:
@kappa95 Remove the rigid joint between base and the center point of the environment and instead make the base component grounded. You can do this by right clicking on the component and selecting it in the context menu. The center point resolves to the root component of your scene which has a weird kind of container nature, and thus behaves oddly when trying to create physical relationships with other parts in the design.

kappa95, 2023-10-26:
It worked!! Thanks! But where is it saved? I don't find any explicit filepath in the code

KyroVibe, 2023-10-26:
The generated mirabuf file is at `%appdata%/Autodesk/Synthesis/Mira/`. Fields (Static exports) are located in the Fields subdirectory and dynamic files exported with the part flag enabled are located at `%appdata%/Autodesk/Synthesis/MixAndMatch/Mira`

PR #950 (2023-11-08): Fixed MacOS Dependencies for Exporter
### Description
Removed the use of `os.__file__` due to new issues with it being frozen in python v3.11 (see #948). In addition, modified the path concatenation.

### Testing Done
- [x] Tested MacOS can install dependencies
- [x] Tested MacOS can install dependencies AGAIN after update
- [x] Tested Windows can install dependencies AGAIN after update

### Notes
This should hopefully resolve #948 but will be difficult to confirm do to the illusive nature of the bug.
closed 2023-10-27 by KyroVibe